hackclub/OnBoard #150: Community call

maxwofford, 2023-07-19:


Leo32345, 2023-07-19:
Time: [Friday, July 28 @ 8pm EST](https://www.timeanddate.com/worldclock/fixedtime.html?msg=OnBoard+Community+Meeting%21&iso=20230728T20&p1=4812&am=30)

About: Hey everyone, recently we've been noticing a lot of great PCB design submissions for [OnBoard](https://github.com/hackclub/Onboard), congrats to everyone who've already made a submission! 🎉
We'll be hosting an OnBoard meeting where YOU can share what you've made so far and meet other Hackclubbers! Even if you haven't designed a PCB yet, you're welcome to join us! 👋

maxwofford, 2023-07-20:
Marketing ideas (copied from Slack)

- [events.hackclub.com](http://events.hackclub.com/)
- [#happenings](https://hackclub.slack.com/archives/C05B6DBN802)
- email to everyone who submitted a PR or forked the repo

maxwofford, 2023-07-20:
It looks like we haven't made a post or announcement about this yet, so I'd propose we run this call this Sunday to give ourselves another 2 days to get attendees.

closed 2024-03-01 by maxwofford