hackclub/OnBoard #163: regarding the parts on the pcb

BadPotato1007, 2023-08-08:
hey there! I was planning to order a pcb from jlc pcb, and just say the MASSIVE production costs that they have. so is it possible that I order the ics and the smd components from somewhere else and assemble the board my self? will the cost of those parts be included in the project funds?

BadPotato1007, 2023-08-08:
the parts are a drones flight controller that I have designed, and the ics are very cheap here... I will al be exempted or a huge chunk of import duty that I would otherwise have to pay...

karmanyaahm, 2023-09-14:
Hi! Sorry for the late reply, but no, the only vendors supported by OnBoard are in [vendors.md](https://github.com/hackclub/OnBoard/blob/main/docs/VENDORS.md). Please ask any further questions on the #onboard channel on Slack.

closed 2023-09-14 by karmanyaahm