hackclub/OnBoard #233: Slack channel broken

kcoderhtml, 2023-10-24:
When I tried to use the link for the #onboarding channel it gave me this error: 
[image]
I have tried four times over the last 3 days.

maxwofford, 2023-10-25:
@kcoderhtml thanks for flagging this! I'm looking into it now.

maxwofford, 2023-10-25:
Hmmmm, I just went through the links in the README and they seem to have worked. @kcoderhtml, could you please send the URL linked for you? Where did you find the link?

kcoderhtml, 2023-10-25:
I found the link on the README: https://hackclub.slack.com/archives/C056AMWSFKJ. It only seems to happen when I'm logged in to slack.

maxwofford, 2023-10-25:
Oh, interesting! Could you try using this link to test if this works better? https://app.slack.com/client/T0266FRGM/C056AMWSFKJ

kcoderhtml, 2023-10-25:
I think I found the problem. I hadn't completed the signup process in slack.

closed 2023-10-25 by kcoderhtml