hackclub/OnBoard #1059: Change ordering from JLCPCB guide

AverseABFun, 2024-09-02:
After following it, I noticed that after checking out it clears your cart and you now need to go to https://jlcpcb.com/user-center/orders to view the order. Also, various settings have been changed/removed and so it overall needs to be updated.

blazecoding2009, 2024-09-02:
What are you trying to accomplish with this PR?

AverseABFun, 2024-09-03:
> What are you trying to accomplish with this PR?

This is an issue

LimesKey, 2024-09-03:
Are you talking about this doc [Ordering From JLCPCB](https://github.com/hackclub/OnBoard/blob/main/docs/ordering_from_JLCPCB.md)? The link, https://jlcpcb.com/user-center/orders/, appears to be in that guide and works as expected. What various/settings need to be updated; everything looks the same or close enough for the guide?

You're free to make a PR and update the ordering from the JLCPCB documentation guide.

AverseABFun, 2024-09-03:
Oh whoops, looks like I misread it. Closing this issue now

closed 2024-09-03 by AverseABFun