hackclub/OnBoard #1324: Will PCBX be added as a manufacturer?

KOEGlike, 2024-11-18:
They have a deal right now that you get 10 free PCBs and PCBA, you just have to pay for the components

LimesKey, 2024-12-16:
We can only approve vendors that...

* Only fulfill PCB fabrication (no general vendors like Mouser, Digikey, etc.)
* Provide itemized receipts
* Work with HCB cards

At this time we aren't accepting vendors that provide extra services. Does PCBX follow this criteria?

maxwofford, 2025-01-04:
@LimesKey I'm checking them out now and it looks good! @KOEGlike we'll need your help to detect their merchant ID. Can you use your grant card with them? The charge will be declined, but it'll give us the details we need to add the vendor to the allowed merchant list.

KOEGlike, 2025-01-04:
Nice! I will try to make a payment

KOEGlike, 2025-01-12:
Realized that they only accept bank transfers, they don't have a payment processor

closed 2025-01-12 by KOEGlike