hackclub/OnBoard #1510: Insufficient Instructions regarding payment

MUmarShahbaz, 2025-03-10:
I just bought a PCB with the grant money and here are the following things that I believe should be added to the `README.md`

1. What counts as a receipt?
   - I used JLCPCB and I have 4 options to choose from for uploading the receipt
     - The Invoice : Just the bill, does not show any proof of payment
     - The Sales Contract: Available only after the payment
     - The order details page: Shows the current status of the board along with the payment date, private info is already hidden by default by JLCPCB so there is no need to go out of the way to remove it.
     - Any Physical Receipts that arrive with the order (if they do)

2. What to do with the remainder of the money?
   - Can I keep it?
   - If I am allowed to keep it, where else am I allowed to use it?
     Perhaps some standalone components irrelevant to the PCB?
     Or do I have a complete freedom with the remainder?

MUmarShahbaz, 2025-03-11:
Also one more quick question. Since we are being given a virtual card, if we have the option to pay for customs fee using it, can we??

tobycm, 2025-03-29:
doesnt they recommend paying custom fee using the global direct line shipping option? hack club cant cover custom fee, it is in the README

MUmarShahbaz, 2025-03-29:
I'm under the impression that it's a technical issue rather than we are not allowed to. So if there is a way to do it, can we?

The virtual card only allows payment to a select few people. If they awere to add DHL to it, then we could pay the customs fees easily.

tobycm, 2025-03-30:
Hack Club is poor, i dont think they can cover customs. Thats why they recommend us to pick global line to handle customs for us and save money.

tobycm, 2025-03-30:
Also for the remainder, for future boards you can pay for them.

tobycm, 2025-03-30:
But yeah i think these info should definitely be added to the README